AutomotiveAIChallenge/aichallenge2022-sim #2: 動作確認について

DAIKI-NUMATA, 2022-11-25:
Autoware、AWSIMの動作確認時の箇所での質問になります。

自己位置の推定ができません。

AutowareStateの状態がINTとなっており、2D Pose Estimate をクリックし、map上で指定しても何も起こりません。
（ドラックすると小さなベクトルのようなものは出るが、消えてしまう）

環境構築のところで手順ミスがあるのでしょうか？
よろしくお願いいたします。

hir-yk, 2022-11-28:
すでにROS2環境が導入されていて ROS_DOMAIN_ID が設定されていたりしませんか？

DAIKI-NUMATA, 2022-11-28:
お忙しいところご回答ありがとうございます。

一応今回使用しているSSDは、OSをインストールするところから行っているため、ROS2環境が入っていたということはないと思います。
しかし手順の『rocker』をインストールするためにROS Noetic をインストールしています。

 ROS_DOMAIN_ID の確認方法を教えていただきたいです。

また、Autoware初期起動時はmapも表示されていたのですが、本日起動してみたところmapが表示されず、
AutowareのDisplaysパネルの『Global Status : Warn』となっています。
（No tf data. Actual error : Frame [map] does not exist）

大変お手数ですがお力添え願います。

↓【Autoware起動時のコンソール】----------------------------------------------------------------------------------------
[web_server.py-51] [INFO] [1669629769.550139605] [web_server]: service not available, waiting again... (/api/interface/version
[rviz2-63] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-63]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.4/src/buffer_core.cpp
----------------------------------------------------------------------------------------

seigot, 2022-11-28:
お疲れ様です。私の方でも以下のREADMEを参考に環境構築＆engageまで実行を試みたのですが、自己位置推定ができていなさそうな状態になりました。色々確認していたところ本issueを見つけてコメントさせて頂きました。
RVIZの画像を添付いたします。他にも何か添付したほうが良いログ等ありましたら添付するようにいたします。
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/main/README.md

[image]

Reee009876, 2022-11-28:
お世話になっております。

確認いたしますので、少々お待ちください。

DAIKI-NUMATA, 2022-11-29:
OSを入れ直すところから再度環境構築を試みたところ正常な状態になり動作確認が取れました。
ありがとうございました。

ちなみに

DAIKI-NUMATA, 2022-11-29:
お世話になっております。OSを入れ直すところから環境構築を再度行い、動作確認することができました。
ありがとうございました。

ちなみに自己位置推定はAutoware起動したコンテナに入り AWSIM を 起動したことで推定できるようになりました。

seigot, 2022-11-30:
ありがとうございます。上記の方法で自己位置推定まで動作することができました。
（Autoware起動したコンテナに入り AWSIM を 起動）
しかし2D map poseを押してもengageの実行待ちにすることができておりません、、
こちらの方法ではengageも実行できましたでしょうか？？また、画像を添付します。もし何か確認が必要なことがあれば確認進めたいと思いますので回答いただけますと幸いです。

>OSを入れ直すところから環境構築を再度行い、動作確認することができました。
>ありがとうございました。
>ちなみに自己位置推定はAutoware起動したコンテナに入り AWSIM を 起動したことで推定できるようになりました。

[image]

DAIKI-NUMATA, 2022-11-30:
お世話になっております。

ホストOS内で先にAWSIMを起動し、autowareのローンチを起動しましたが「AutowareStatePanel」はすべて「INIT」となっており、うまくAWSIMと通信できていないような状況です。

グラフィックボード：GEFORCE RTX 3080 12GB （ZOTAC GAMING）

log は /var/log/syslog を提出すればよろしいでしょうか？

よろしくお願いいたします。

Reee009876, 2022-12-01:
>map上で指定しても何も起こりません。
とのことなので，三次元地図などは表示されるということでよろしいでしょうか？？
地図は表示されるけどもLiDARの点群などが表示されず，INT表示となっている．

DAIKI-NUMATA, 2022-12-01:
はい、地図は表示されています。ただ表示されているだけで、車両はない状態になります。
AutowareStatePanelの表示はINITです。

AWSIMを同一コンテナ内で起動した時は出て、ゴールまでの一通りの動作確認まではできていました。

お手数ですがよろしくお願いいたします。

Reee009876, 2022-12-01:
ご確認いただきまして、ありがとうございます。

ログですが、autowareを起動した際に下記のような表示がされると思いますので、そちらの提出をお願いしたいです。
`[INFO] [launch]: All log files can be found below /home/user/.ros/log/2022-12-01-23-36-34-273103-***********
`

DAIKI-NUMATA, 2022-12-01:
log提出します。

_エラー文なども出ていない状況でしょうか？_
→エラーも何も出ていないです。

よろしくお願いいたします。

↓【Autoware起動時log】(Awsim起動済み→Autoweare起動)
[log_autoware.log](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/files/10135762/log_autoware.log)

kasecato, 2022-12-03:
### TL;DR
AWSIM を起動するホスト OS 側に ROS2 Galactic をインストールします。
- https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html

### 推測
「コンテナ内で AWSIM を起動した場合に動作した」という趣旨のコメントがありました。おそらく `autoware-universe-cuda` コンテナ側と同じバージョンの ROS2 Galactic をホスト側にインストールすることで動作するのでは思い試してみました。

#### As-Is
- ROS2 Galactic 未インストール
- `AWSIM` が rqt_graph に表示されない
- AWSIM と Autoware が会話していないように見え，車両が動かない
[image]

#### To-Be
- ROS2 Galactic インストール済み
- `AWSIM` が rqt_graph に表示される
- AWSIM と Autoware が会話しているように見え，車両が動いた
[image]

### ただし…
自分の環境では AWSIM だけがよく異常終了します。他の GUI アプリも巻き込んでハングして，マウスも動かなくなり 1 分程度待つと AWSIM が落ちて他の GUI アプリが使用できるようになります。AWSIM と Rviz2 が正常に動作するまで何度も起動し続けて漸く動作するようになるまで繰り返しています。

```bash
$ nvidia-smi
| NVIDIA-SMI 515.86.01    Driver Version: 515.86.01    CUDA Version: 11.7     |

$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.5 LTS (Focal Fossa)"

$ cat /proc/cpuinfo | grep -m 1 "model name"
model name	: AMD Ryzen 7 5800H with Radeon Graphics
```

DAIKI-NUMATA, 2022-12-03:
お力添えありがとうございます。

私の環境で正常に動作していることを確認することができました。
大変ありがとうございました。

rocker を install する際に ROS を install となっていたので ROS2 ではなく 
ROS noetic を install していました。

ありがとうございました。

DAIKI-NUMATA, 2022-12-04:
一応 Close とさせていただきます。

closed 2022-12-04 by DAIKI-NUMATA