AutomotiveAIChallenge/aichallenge2022-sim #5: ルートが表示されない件について

Kenzo-Fujisaki, 2022-12-08:
- 動作確認において自己位置推定はできるのですが、2D Goal Poseを道路においた際にルートを表示してくれない問題が起きています。解決方法はありますでしょうか。動作環境はPC2台でどちらともスペックは満たしております。

[image]

Reee009876, 2022-12-08:
お世話になっております。

解決のため、下記のご対応をお願いできないでしょうか？
・Autowareのログの提出
・Autowareのログ内に”Not Covered”がないかの確認
・自己位置推定後、数分程度様子見

よろしくおねがいします。

Kenzo-Fujisaki, 2022-12-09:
返答ありがとうございます。対応を行った結果を以下に記述します。
> Autowareのログの提出
```
$ ros2 launch aichallenge_launch aichallenge.launch.xml
[INFO] [launch]: All log files can be found below /home/tsukuba2/.ros/log/2022-12-09-16-11-29-014475-tsukuba2-desktop-68864
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/system/system_monitor/cpu_monitor' in container '/system/system_monitor/system_monitor/system_monitor_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/sensing/lidar/right/pointcloud_distortion_relay' in container '/sensing/lidar/right/pointcloud_preprocessor/velodyne_node_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/perception/occupancy_grid_map/occupancy_grid_map_node' in container '/perception/occupancy_grid_map/occupancy_grid_map_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/perception/object_recognition/detection/voxel_based_compare_map_filter' in container 'pointcloud_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/perception/obstacle_segmentation/crop_box_filter' in container 'pointcloud_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/control/external_cmd_converter' in container '/control/control_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/sensing/lidar/concatenate_data' in container 'pointcloud_container'
[INFO] [score-1]: process started with pid [68896]
[INFO] [component_container_mt-2]: process started with pid [68898]
[INFO] [robot_state_publisher-3]: process started with pid [68900]
[INFO] [component_container_mt-4]: process started with pid [68902]
[INFO] [ad_service_state_monitor-5]: process started with pid [68904]
[INFO] [aggregator_node-6]: process started with pid [68906]
[INFO] [system_error_monitor-7]: process started with pid [68908]
[INFO] [emergency_handler-8]: process started with pid [68910]
[INFO] [component_container-9]: process started with pid [68912]
[INFO] [map_hash_generator-10]: process started with pid [68914]
[INFO] [component_container-11]: process started with pid [68916]
[INFO] [component_container-12]: process started with pid [68918]
[INFO] [component_container-13]: process started with pid [68920]
[INFO] [imu_corrector-14]: process started with pid [68922]
[INFO] [vehicle_velocity_converter-15]: process started with pid [68924]
[INFO] [map_height_fitter-16]: process started with pid [68926]
[INFO] [pose_initializer_node-17]: process started with pid [69081]
[INFO] [automatic_pose_initializer-18]: process started with pid [69084]
[INFO] [ndt_scan_matcher-19]: process started with pid [69088]
[INFO] [gyro_odometer-20]: process started with pid [69105]
[INFO] [ekf_localizer-21]: process started with pid [69152]
[INFO] [stop_filter-22]: process started with pid [69154]
[INFO] [twist2accel-23]: process started with pid [69156]
[INFO] [localization_error_monitor-24]: process started with pid [69158]
[INFO] [component_container_mt-25]: process started with pid [69183]
[INFO] [component_container-26]: process started with pid [69186]
[INFO] [shape_estimation-27]: process started with pid [69216]
[INFO] [detected_object_feature_remover-28]: process started with pid [69222]
[INFO] [detection_by_tracker-29]: process started with pid [69224]
[INFO] [lidar_centerpoint_node-30]: process started with pid [69258]
[INFO] [obstacle_pointcloud_based_validator_node-31]: process started with pid [69275]
[INFO] [object_association_merger_node-32]: process started with pid [69277]
[INFO] [object_association_merger_node-33]: process started with pid [69279]
[INFO] [object_lanelet_filter_node-34]: process started with pid [69286]
[INFO] [multi_object_tracker-35]: process started with pid [69291]
[INFO] [map_based_prediction-36]: process started with pid [69322]
[INFO] [traffic_light_map_based_detector_node-37]: process started with pid [69359]
[INFO] [component_container_mt-38]: process started with pid [69385]
[INFO] [traffic_light_map_visualizer_node-39]: process started with pid [69387]
[INFO] [mission_planner-40]: process started with pid [69389]
[INFO] [goal_pose_visualizer-41]: process started with pid [69391]
[INFO] [scenario_selector-42]: process started with pid [69411]
[INFO] [external_velocity_limit_selector-43]: process started with pid [69414]
[INFO] [motion_velocity_smoother-44]: process started with pid [69506]
[INFO] [component_container_mt-45]: process started with pid [69508]
[INFO] [rtc_auto_mode_manager_node-46]: process started with pid [69519]
[INFO] [component_container_mt-47]: process started with pid [69542]
[INFO] [component_container-48]: process started with pid [69572]
[INFO] [planning_error_monitor-49]: process started with pid [69623]
[INFO] [component_container-50]: process started with pid [69626]
[INFO] [component_container_mt-51]: process started with pid [69634]
[INFO] [web_server.py-52]: process started with pid [69653]
[INFO] [initial_pose_adaptor-53]: process started with pid [69660]
[INFO] [routing_adaptor-54]: process started with pid [69664]
[INFO] [awapi_awiv_adapter-55]: process started with pid [69693]
[INFO] [component_container-56]: process started with pid [69714]
[INFO] [component_container_mt-57]: process started with pid [69717]
[... 1279 lines cut ...]
[... cut ...]
[component_container_mt-51] [INFO] [1670569895.111570685] [default_ad_api.node.localization]: service exit: /api/localization/initialize
[component_container_mt-51] status:
[component_container_mt-51]   success: false
[component_container_mt-51]   code: 4
[component_container_mt-51]   message: "NDT align server failed."
[component_container_mt-51] 
[automatic_pose_initializer-18] [INFO] [1670569895.111707081] [localization.util.default_ad_api.helpers.automatic_pose_initializer]: client exit: /api/localization/initialize
[automatic_pose_initializer-18] status:
[automatic_pose_initiali

Kenzo-Fujisaki, 2022-12-17:
本日確認したところ無事走り出しました。
ご対応いただきありがとうございます。

[image]

closed 2022-12-17 by Reee009876