AutomotiveAIChallenge/aichallenge2022-sim #6: Challengeコースのサンプルのエラーと対処法について

kosei1515, 2022-12-09:


Reee009876, 2022-12-09:
お世話になっております。

情報のご共有ありがとうございます。
問題に関してですが、「シミュレータは動作するが、自動運転が開始されない」ということでよろしかったでしょうか？

上記であれば、「環境や負荷によっては自動運転が開始されないことがある」ことを
こちらでも確認しており、現在、原因・解決策を調査しているような状況です。
※チャレンジコース以外でも発生。

大変申し訳ありません。
解決までは、記載頂いている方法で対処いただけると幸いです。

kosei1515, 2022-12-09:
> シミュレータは動作するが、自動運転が開始されない
シミュレータが起動して画面に車とライダーが映っている状態でシミュレータが動作しているのであればその認識で間違いないと思います

> ※チャレンジコース以外でも発生していることを確認しています。
参考までにですが
自分たちのPCではAdvancedコースで同様の現象は見受けられませんでした

Reee009876, 2022-12-09:
> シミュレータが起動して画面に車とライダーのラインが映っている状態でシミュレータが動作しているのであればその認識で間違いないと思います
こちらですが、[add new panel]からManualContllerを追加していただいて、手動操作をしての確認をお願いできないでしょうか？
[image]

> 自分たちのPCではAdvancedコースで同様の現象は見受けられませんでした
追加情報ありがとうございます。

kosei1515, 2022-12-09:
やってみました　こちら動画です
おそらくシミュレータは動作していると思います

[Screencast from 12-09-2022 07:02:02 PM.webm](https://user-images.githubusercontent.com/81236654/206676996-f2dee84f-a3bc-46a9-8845-99cc0f8f1d56.webm)

Reee009876, 2022-12-12:
お世話になっております。

自己位置がうまくいかない問題に対応した
docker imageを公開しました。
一度、こちらをお試しいただけないでしょうか？

https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/issues/1

Reee009876, 2023-01-15:
こちら、修正版のアップロードが完了しておりますため、
クローズとさせていただきます。

また、疑問などがございましたら
再オープンをお願いします。

closed 2023-01-15 by Reee009876