AutomotiveAIChallenge/aichallenge2022-sim #10: docker pull でファイルをダウンロードが完了できない

fukufukufukukawa, 2022-12-16:
お世話になります。

dockerインストール後、
docker pull ghcr.io/automotiveaichallenge/aichallenge2022-sim/autoware-universe-cuda:latest

を実施していますが、
7.145GBのファイルの転送だけが遅く（１時間に1GB程度）毎回接続リセットとなります。
今週月曜日から、時間帯を変更しながら実施したのですが、いずれも成功しておりません。
対処方法を教えてください。

接続は光で、PCとはWIFIで接続しております。

846c0b181fff: Pull complete 
e41b30a1a126: Pull complete 
db29ca785f15: Pull complete 
b230327fb394: Pull complete 
4f4fb700ef54: Pull complete 
bea0778d4fa4: Pull complete 
385252337af6: Downloading  4.884GB/4.884GB
c427542782a4: Download complete 
1d4f9c179d60: Download complete 
3f1529196284: Download complete 
95ee91f8c5e2: Download complete 
450fe2bc00d6: Download complete 
01b965867c6e: Download complete 
ed8904857edd: Download complete 
17288e8a7872: Download complete 
read tcp 192.168.0.26:53304->52.239.170.4:443: read: connection reset by peer

もしくは、
read tcp 192.168.0.26:50010->52.239.170.4:443: read: connection timed out

Reee009876, 2022-12-17:
お世話になっております。

下記のフォルダにイメージをtarにまとめたものをおいてありますので、こちらをご利用ください。
https://drive.google.com/drive/u/2/folders/1VZAcGzcFpOBJlmmybcGau7BaHzZW5Chc

`gzip -d aichallenge2022_sim_autoware_v2.0.tar.gz  `
`docker load  < aichallenge2022_sim_autoware_v2.0.tar`

Reee009876, 2023-01-15:
こちら、修正版のアップロードが完了しておりますため、
クローズとさせていただきます。

また、疑問などがございましたら
再オープンをお願いします。

closed 2023-01-15 by Reee009876