AutomotiveAIChallenge/aichallenge2022-sim #12: 走行タイムの確認ができません

hidechun, 2022-12-18:
お世話になっております。

環境構築に苦戦中の参加者です。
他のISSUEも参考にしながら、チャレンジコースを自走するところまで進めることができました。

しかし、RULE.mdのタイムの確認項にある
source /aichallenge/aichallenge_ws/install/setup.bash
ros2 topic echo /score/result

をLocker上で何度打ち込んでも以下のscoreが表示されません。
走行前、走行後に試し、走行後にターミナル表示にscoreが紛れていないかチェックしましたがありませんでした。

他にもお困りの方はいませんでしょうか。

ーーー探したスコア表示ーーー
# score: 164699
# has_finished: true
# has_collided: false
# has_park_failed: false
# check_point_count: 2
ーーーーーー

hidechun, 2022-12-18:
[challange_log.TXT](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/files/10253879/challange_log.TXT)

Reee009876, 2022-12-18:
お世話になっております。

もう一枚、コンテナにアクセスできるコンソールを用意していただき、
そちらの画面でコマンドを打って確認していただいてもよろしいでしょうか？

hidechun, 2022-12-18:
お世話になります。
コンテナにアクセスできるターミナルを用意し、
source /aichallenge/aichallenge_ws/install/setup.bash
ros2 topic echo /score/result
を入力したところ、ゴール到着と同時に結果表示することができました。
ご助言ありがとうございました。

表示された内容
ーーーーーーーーーーーーーーーーーーー
root@hidechun-LAPK43:/autoware# source /aichallenge/aichallenge_ws/install/setup.bash

root@hidechun-LAPK43:/autoware# ros2 topic echo /score/result
score: 149286
has_finished: true
has_collided: false
has_park_failed: false
has_exceeded_speed_limit: false
check_point_count: 2
---

closed 2022-12-18 by hidechun