AutomotiveAIChallenge/aichallenge2022-sim #14: test

seigot, 2022-12-19:
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/issues/6

4.RvizのタブにあるPanelからadd new Panelを開き、AutowareStatePanelを追加
5.自己位置推定ができていることを確認
6.正しく推定できていなければ、タブにある2D Pose Estimateを選択し、実際の車両の位置をドラッグで指定

[image]

> [component_container-63] [WARN] [1671466577.943766958] [control.trajectory_follower.lane_departure_checker_node]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[component_container-63] [INFO] [1671466577.943818324] [control.trajectory_follower.lane_departure_checker_node]: waiting for self pose...
[motion_velocity_smoother-57] [WARN] [1671466578.116429485] [planning.scenario_planning.motion_velocity_smoother]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[motion_velocity_smoother-57] [INFO] [1671466578.116466589] [planning.scenario_planning.motion_velocity_smoother]: waiting for self pose...
[aggregator_node-20] [ERROR] [1671466578.133841860] [generic_analyzer_base]: Attempting to convert 255 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466578.133877884] [generic_analyzer_base]: Attempting to convert 224 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466578.133888595] [generic_analyzer_base]: Attempting to convert 52 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466578.133895267] [generic_analyzer_base]: Attempting to convert 111 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466578.133901907] [generic_analyzer_base]: Attempting to convert 64 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466579.133835107] [generic_analyzer_base]: Attempting to convert 255 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466579.133861886] [generic_analyzer_base]: Attempting to convert 224 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466579.133896271] [generic_analyzer_base]: Attempting to convert 52 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466579.133901542] [generic_analyzer_base]: Attempting to convert 111 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466579.133905712] [generic_analyzer_base]: Attempting to convert 64 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466580.133823810] [generic_analyzer_base]: Attempting to convert 255 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466580.133853436] [generic_analyzer_base]: Attempting to convert 224 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466580.133861150] [generic_analyzer_base]: Attempting to convert 52 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466580.133865520] [generic_analyzer_base]: Attempting to convert 111 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466580.133869832] [generic_analyzer_base]: Attempting to convert 64 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[component_container-61] [INFO] [1671466580.842849443] [planning.scenario_planning.parking.costmap_generator]: waiting for initial pose...
[aggregator_node-20] [ERROR] [1671466581.133865692] [generic_analyzer_base]: Attempting to convert 255 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466581.133898952] [generic_analyzer_base]: Attempting to convert 224 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466581.133907304] [generic_analyzer_base]: Attempting to convert 52 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466581.133911980] [generic_analyzer_base]: Attempting to convert 111 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466581.133916517] [generic_analyzer_base]: Attempting to convert 64 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[component_container_mt-60] [WARN] [1671466581.802123895] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[component_container_mt-60] [INFO] [1671466581.802173966] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: waiting for self pose...
[aggregator_node-20] [ERROR] [1671466582.133837585] [generic_analyzer_base]: Attempting to convert 255 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466582.133868636] [generic_analyzer_base]: Attempting to convert 224 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466582.133876974] [generic_analyzer_base]: Attempting to convert 52 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466582.133881582] [generic_analyzer_base]: Attempting to convert 111 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-20] [ERROR] [1671466582.133885988] [generic_analyzer_base]: Attempting to convert 64 into DiagnosticLevel.
[aggregator_node-20]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[component_container-63] [WARN] [1671466582.943909461] [control.trajectory_follower.lane_departure_checker_node]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[component_container-63] [INFO] [1671466582.943947963] [control.trajectory_follower.lane_departure_checker_node]: waiting for self pose...
[motion_velocity_smoother-57] [WARN] [1671466583.116549456] [planning.scenario_planning.motion_velocity_smoother]: failed to get trans

seigot, 2022-12-20:
ここでautowareが正しいかどうか確認します

> cd ./aichallenge2022-sim
rocker --nvidia --x11 --user --net host --privileged --volume autoware:/aichallenge -- ghcr.io/automotiveaichallenge/aichallenge2022-sim/autoware-universe-cuda:latest

Reee009876, 2022-12-20:
以下を打っていただけますか？
`ros2 topic echo /sensing/lidar/top/pointcloud_raw`

seigot, 2022-12-26:
遅くなりすみません、上記記載の状態でコメント頂いたコマンドを実行したところ以下のログが出力されました。
何かわかりましたらコメントいただけますと幸いです

```
ubuntu@ubuntu-FRONTIER:~$  ros2 topic echo /sensing/lidar/top/pointcloud_raw
WARNING: topic [/sensing/lidar/top/pointcloud_raw] does not appear to be published yet
Could not determine the type for the passed topic
```

Reee009876, 2022-12-29:
ありがとうございます。

ホスト側とコンテナ側でトピック通信ができていないように見受けられます。
AWSIMを起動しておられるホスト側に、ROS2はインストールされていらっしゃいますか？
ご確認お願いします。

seigot, 2022-12-30:
ご回答ありがとうございます、ホスト側へROS2をインストールができておりませんでした、、
解決しましたのでクローズさせて頂きます。初歩的な質問ですみません。

closed 2022-12-30 by seigot