AutomotiveAIChallenge/aichallenge2022-sim #15: オンライン環境にソースコードをアップロードするためのURLについて

seigot, 2022-12-27:
以下に表題の記載がありますが、これはどちらのURLになりますでしょうか（既出でしたらすみません）
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/main/RULE.md#%E7%AB%B6%E6%8A%80%E5%85%A8%E4%BD%93%E3%81%AB%E9%96%A2%E3%81%99%E3%82%8B%E3%83%AB%E3%83%BC%E3%83%AB

> 競技全体に関するルール
[AWSIM](https://github.com/tier4/AWSIM)(自動運転向けシミュレータ)上でのシナリオを走破できるソースコードを[Autoware](https://github.com/autowarefoundation/autoware/tree/awsim-stable)を活用して作成いただきます。
参加者は、与えられたシナリオをクリアできるようソースコードを作成して、まずはローカル環境で検証をします。
オンライン環境にソースコードをアップロードすることで、オンライン上でシミュレーションが実施されて、タイムが出ます。最後にアップロードされたソースコードのシミュレーション結果のタイムに基づいて順位を決定します。

Reee009876, 2022-12-29:
お世話になっています。

オンライン環境へのご案内はまだ行っておらず、
現在、来月上旬でのご案内を予定しております。

よろしくおねがいします。

seigot, 2023-01-10:
公開ありがとうございました！

closed 2023-01-10 by seigot