AutomotiveAIChallenge/aichallenge2022-sim #16: 他チームと交流できる場はあるかどうか質問させてください。

seigot, 2022-12-27:
他チームと交流できる場はあるかどうか質問させてください。

本大会が目指している以下の姿に非常に共感しております。
自動運転技術はこれからの自動車業界（これに留まらず他業界にも）に大きな影響を与えるものであり、その技術の獲得はおそらくどの組織に属するものでも有意義なものになると思います。学習機会を提供いただきありがとうございます。

> https://www.jsae.or.jp/jaaic/about.php
> Japan Automotive AI Challenge　自動運転AIチャレンジとは？
本大会は、CASE、MaaSと呼ばれる新たな技術領域において、これからの自動車業界を牽引する技術者の発掘育成のための新たな取り組みとして実施しています。

しかしながら元々この業界に関わりが薄かった者にとっては、どの技術も学習ハードルが高く取り組みが困難なものになる傾向がありそうです。詳しい方にアドバイスを頂けると助かるのですが、これもなかなか難しいとは思っております。
そのため、例えば同じ悩みを持つチーム同士で交流できるような場があればいい落としどころとなるのではと思っております。
（と書きつつもしすでに存在しておりましたら申し訳ございません。）

ncbwct1, 2022-12-28:
自動運転AIチャレンジ事務局でございます。
この度はご参加いただき誠にありがとうございます。
また、本大会の理念にご賛同いただいた上でご意見頂戴し大変恐縮に存じます。

大変申し訳ないのですが、現在チーム同士で交流できる場というのは運営側からは提供しておりません。
ご指摘の通り自動運転に係る諸技術は環境構築や学習のハードルが高く、他の参加者様からもその旨ご意見いただいております。
事務局としてどのような形で参加者様の学習環境を整えるかというのは目下検討中でございます。
seigot様のチーム同士の交流の場というのは大変有用な方法かと存じますので、今大会は難しいと思いますが次回以降Slack等においてそのような場を設けさせて頂きたいと思います。

解決策をご提示できず大変申し訳ございません。
ＡＩチャレンジは来年度も年2回のペースで開催予定ですので、引き続きご参加いただけますと幸いです。

seigot, 2022-12-29:
ご回答ありがとうございます。次回以降ご検討いただける旨承知いたしました。たいへんありがたいです。
ちなみにですが（非公式にはなりますが)雑談を目的としたslackであれば有志で作成可能とは思うのですが、このあたりはなにか制約などございますでしょうか。

ncbwct1, 2023-01-05:
回答遅くなり大変申し訳ございません。
有志でのslack等の利用に制約などは設けておりません。
ご活用いただければ幸いです。

closed 2023-02-05 by seigot