AutomotiveAIChallenge/aichallenge2022-sim #19: オンライン評価環境におけるチャレンジ コースのスコアの目安について

kasecato, 2023-01-07:
- `環境:` オンライン評価環境
- `コース:` チャレンジ コース
- `ソースコード:` サンプル コードをそのまま ac1541f77ff530e1498bd2b31b3bade782181a71
- `スコア`: 2023-01-07 07:26  採点完了  64.77秒 <- 速すぎる？

### 質問
オンライン環境のチャレンジ コースではサンプル コードの状態でスコアがおおよそ何秒台か目安はありますでしょうか？

### 背景
上記条件下で試したところ 64 秒台で採点完了となり，異常に高速だと思い質問しました。
また，オンライン環境とローカル環境には内容の差分があることは `RULE.md` に以下の通り記載されていることは把握しております。

> オンライン採点環境・ローカル環境両方で以下の内容は共通していますが、イベントの順番、車両の位置等細かい内容は異なります。
> 各イベントについて記載している走破方法以外の経路・方法で走破した場合、オンライン採点環境では動作しない可能性があります。

しかし，ローカル環境ではサンプル コードで約 160 秒台だったコースが 64 秒台になるのか気になったため，お問い合わせしました。

- RULE.md, 「シナリオ」, https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/release_20221229/RULE.md#%E3%82%B7%E3%83%8A%E3%83%AA%E3%82%AA

hidechun, 2023-01-08:
横から失礼致します。
お世話になっております。

当方のPC上でAWSIMを用いたシミュレーションで160秒台
本日オンライン環境にアップロードしたところ、以下の結果になりました。
高速すぎることに加え、スコアのばらつきもかなり大きく感じております。

2023-01-08 12:45 | 採点完了 | 77.85秒
2023-01-08 11:36 | 採点完了 | 75.85秒
2023-01-08 10:17 | 採点完了 | 64.90秒

考察ございましたらご教示いただけると幸いです。

hidechun, 2023-01-09:
参考情報です。本日も試してみました。

2023-01-09 15:50 | 採点完了 | 76.24秒
2023-01-09 14:08 | 採点完了 | 102.27秒
2023-01-09 08:23 | 採点完了 | 129.16秒

kasecato, 2023-01-09:
オンライン シミュレータで5回の平均計算に失敗した場合の外れ値が含まれているような挙動に見えるという印象です。

Reee009876, 2023-01-14:
おまたせしてしまい、大変申し訳ありません。

オンライン環境においてランキングのスコアがばらける問題について、修正が完了したため、
再度実行のほど、お願いいたします、

kasecato, 2023-01-14:
@Reee009876 san,
修正を確認しました。スコアが安定しているため問題ありません。
また，サンプルのソースコード 130 秒台と把握しました。

@hidechun san,
もし問題がなければ Close したいと思います。

hidechun, 2023-01-14:
アップロード後動作確認し、130秒近辺の出力になっていることを確認しました。
修正前は１時間かかっていたシミュレーション完了までの時間ですが、本日は４５分で完了致しました。
2023-01-14 13:29 | 採点完了 | 129.86秒

また、過去のアップロード分の出力結果も修正され、安定しています。
2023-01-08 12:45 | 採点完了 | 129.75秒
2023-01-08 11:36 | 採点完了 | 126.41秒
2023-01-08 10:17 | 採点完了 | 129.79秒

2023-01-09 15:50 | 採点完了 | 127.06秒
2023-01-09 14:08 | 採点完了 | 127.84秒
2023-01-09 08:23 | 採点完了 | 129.16秒

@Reee009876  さま　
AWSIMの実行時間よりは早い１３０秒となっておりますが、修正後に想定された出力時間となっておりますか？

@kasecato さま
想定通りであれば、Close頂いて良いと思います。

kasecato, 2023-01-15:
`結果なし` からしばらくして `採点完了` になるケースがあるようです。
別件のため他の issue とします。

kasecato, 2023-01-19:
@Reee009876 san
オンライン評価環境のチャレンジ コースに「[駐車しようとしている車両の回避](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/main/RULE.md#%E9%A7%90%E8%BB%8A%E3%81%97%E3%82%88%E3%81%86%E3%81%A8%E3%81%97%E3%81%A6%E3%81%84%E3%82%8B%E8%BB%8A%E4%B8%A1%E3%81%AE%E5%9B%9E%E9%81%BF)」は本当に含まれていますでしょうか？

Reee009876, 2023-01-20:
お世話になっております。

@kasecato さま
オンライン環境でのシナリオ中には、ローカル環境と少し動きが異なる「[駐車しようとしている車両の回避](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/main/RULE.md#%E9%A7%90%E8%BB%8A%E3%81%97%E3%82%88%E3%81%86%E3%81%A8%E3%81%97%E3%81%A6%E3%81%84%E3%82%8B%E8%BB%8A%E4%B8%A1%E3%81%AE%E5%9B%9E%E9%81%BF)」が含まれております。

なにかお気づきの点ががあれば、教えていただけると幸いです。
よろしくおねがいします。

kasecato, 2023-01-20:
@Reee009876 san
承知しました。ご確認ありがとうございます。

closed 2023-01-20 by kasecato