AutomotiveAIChallenge/aichallenge2022-sim #20: ローカル環境評価のスコア算出完了時間について

hidechun, 2023-01-08:
お世話になっております。

1月6日に案内のあったDockerイメージ(v3.1)や大会用データを更新し、
以下リンクにあるオンライン・ローカル２パターンのスコア算出をサンプルコードで実施致しました。
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim

当方ローカル評価環境でのスコア算出ができない状況です。
しかし、オンライン評価環境ではスコア算出出来たため、生成ファイルaichallenge_submit.tar.gzに問題は無いと思われます。

一方で、オンライン環境の採点結果を得るのに約１時間要したため、
ローカル環境でのスコア算出も相当な時間待つ必要があるのではないかとも考えており、
ローカル評価環境でのスコア算出出来ない状態がローカル環境の問題なのか、
スコア算出の待ち時間が足りていないのか判断出来ない状況です。

事務局ならびに大会参加者の方でローカル環境にてスコア算出出来た方がいれば、完了時間を共有いただけませんでしょうか。
因みに当方の現況は以下となっております。

　１：オンライン評価環境にソースコードを提出する際の手順におけるローカル評価環境でのスコア算出
　	以下コマンドにてスコア算出を試みました、1時間以上経過しても
　	evaluation/output以下に結果(score.json)が生成されず、フォルダが空状態です。
　
　２：オンライン評価環境におけるのスコア算出
　	オンラインへ　aichallenge_submit.tar.gz　をアップロード後
　	・ロード中状態：約20分
　	・次に、シミュレータテスト中状態：約55分
　	・を経てweb上に採点結果が表示されました。
　	本日３回同一ファイルをアップロードしましたが、ほぼ同様の時間で採点結果を表示しました。
　	結果の妥当性（AWSIM環境より結果時間が早すぎる・結果時間がばらつく）については別ISSUEに書き込みしています。
　	https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/issues/19
　	
　ローカル評価環境の詳細
　以下のコマンドでaichallenge_submit.tar.gzの生成と、ローカル環境でのスコア算出を試みております。
　	cd aichallenge2022-sim/evaluation
	sh create_submit_tar.sh
	sudo sh build.sh Challenge
	sudo sh run.sh
　
　また、コマンド入力後数十分放置した際のLogを添付致します：challange_log2.txt
　*Logの詳細確認が進んでおらずでして、凡ミスでしたら誠に申し訳ありません。　
　
　当方のPC-SPEC
　Hardware
　	・CPU:i7-11800H
　	・RAM:DDR4 32GB
　	・ROM:SSD 512GB
　	・GTX-3060laptop 6GB
　Software
　	・Ubuntu 20.04
　	・Ros2 Galactic

hidechun, 2023-01-08:
[challange_log2.txt](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/files/10368498/challange_log2.txt)

hidechun, 2023-01-08:
間違えてIssueCloseしてしまっていたので、Reopenにさせていただきます。

Reee009876, 2023-01-10:
お世話になっております。

申し訳ありません。
確認しますので、少々お時間をいただきたいです。
また、どのような手順で実行されたかを、詳細に教えていただけないでしょうか？

hidechun, 2023-01-10:
夜分遅くのご確認、ご返信ありがとうございます。
手順について共有させていただきます。
もっと詳細な手順の説明が必要、Logや画面録画など追加情報が必要でしたら対応致します。

■手順・実行の時系列情報
1:12/12日にアナウンスのあったDockerイメージと大会用データを1/6日まで使用しておりました
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim

2:1/6日夜にアナウンスのあったDockerイメージに差し替え（tarデータを使用）
>※上記の方法では長時間かかってしまう方・タイムアウトしてしまう方↓
>[こちら](https://drive.google.com/file/d/145HyoeXye_bbdT6tOVVCvbSM1MTm2CKI/view?usp=share_link)に、イメージをtarにまとめたものを置きましたので、下記コマンドよりご利用ください。
>docker load < aichallenge2022_sim_autoware_v3.1.tar.gz
これを実施し、Dockerデータを差し替えました。

3:1/6日に大会用データを差し替え
homeディレクトリのaichallenge2022-simを削除した後、以下コマンドで再ダウンロードを実施
sudo apt install -y git-lfs
git lfs clone https://github.com/AutomotiveAIChallenge/aichallenge2022-sim

4:サンプルファイルの再ビルドを行い、AWSIM環境上でChallangeコースの完走とterminalを用いた採点ができることを確認
ビルドコマンドは以下です
cd /aichallenge/aichallenge_ws
rosdep update
rosdep install -y -r -i --from-paths src --ignore-src --rosdistro $ROS_DISTRO
colcon build

5:以下リンクの
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim
オンライン評価環境にソースコードを提出する際の手順　を参考に立ち上げるターミナルが１つの状態で
以下のコマンドでaichallenge_submit.tar.gzの生成と、ローカル環境でのスコア算出を試みました。
cd aichallenge2022-sim/evaluation
sh create_submit_tar.sh
sudo sh build.sh Challenge
sudo sh run.sh

本Issueに添付のchallange_log2.txtが実行時のTerminal Logです。

以上です。

hidechun, 2023-01-14:
@Reee009876 さま
お世話になっております。

1/12 12:57のリポジトリ更新アナウンスを受け、
Dockerイメージ（tarデータを使用）と大会用データを差し替えして動作確認致しましたが、
evaluation/output以下に結果(score.json)が生成されず、フォルダが空状態です。

challange_log3.txtが実行時のTerminal Logです。
[challange_log3.txt](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/files/10416884/challange_log3.txt)

以上です。

kosei1515, 2023-01-14:
関連しているようなのでこちらで投稿させていただきます

当方でもローカル環境における採点チェックができていません
なお，インストールはDocker経由，コースはAdvancedコースです

以下実行コマンドです
```
cd evaluation
sh create_submit_tar.sh
sh build.sh Advanced
sh run.sh
```

なお，通常時にシミュレータを起動した際に出るログと評価用のもので実行したときのログを比較すると，以下の2つの例のようなエラーが通常時のシミュレータが動作しているときには見られませんでした
```
[aggregator_node-6] [ERROR] [1673683788.749278140] [generic_analyzer_base]: Attempting to convert 111 into DiagnosticLevel.
[aggregator_node-6]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-6] [ERROR] [1673683788.749287799] [generic_analyzer_base]: Attempting to convert 129 into DiagnosticLevel.
[aggregator_node-6]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-6] [ERROR] [1673683789.749218580] [generic_analyzer_base]: Attempting to convert 255 into DiagnosticLevel.
[aggregator_node-6]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
[aggregator_node-6] [ERROR] [1673683789.749289194] [generic_analyzer_base]: Attempting to convert 111 into DiagnosticLevel.
[aggregator_node-6]     Values are: {0: OK, 1: Warning, 2: Error, 3: Stale}
```
```
[motion_velocity_smoother-50] [INFO] [1673684928.360934509] [planning.scenario_planning.motion_velocity_smoother]: waiting for self pose...
[planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
```
誠に勝手な考察ですが，Autowareとシミュレータの間の接続がうまく行かずエラーが出ているのではないかと思っています．参考程度に報告させていただきました．
以下それぞれのログファイルです
[Advanced_local_normal.txt](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/files/10416921/Advanced_local_normal.txt)

[Advanced_eval_log_1.txt](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/files/10416943/Advanced_elav_log_1.txt)

[Advanced_elav_log_2.txt](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/files/10416955/Advanced_elav_log_2.txt)

Reee009876, 2023-01-15:
お世話になっております。
ご返事が遅くなってしまい、大変申し訳ありません。

ご指摘の通り、AWSIMとの連携がされておらず
自動運転が開始されていない状況です。
（原因：AWSIM起動がされていない）

run.shを叩く前に、AWSIMを起動していただけないでしょうか？
明日、マニュアルは修正いたします。

kasecato, 2023-01-15:
Docker images を 3.1 に更新後から最初の自車両の位置が設定されず，initializing の状態で `e2e_simulator.launch.xml` が動作しません。

[image]

`planning_simulator.launch.xml` に切り替えて `2D Pose Estimate` を実行すると自己位置が [sample.cpp の初期位置](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/main/autoware/aichallenge_ws/src/aichallenge_submit/sample_code_cpp/src/sample.cpp#L115-L125)に設定されます。

hidechun, 2023-01-15:
@Reee009876 さま

1/14日版のDockerイメージ（tar版）に差し替え、
run.shを叩く前に、AWSIMを起動したところ、AWSIM上での自車走行と走行完了後
evaluation/output以下に結果(score.json)が生成されました。

当方環境では、何度か試すと走行できない状態が発生することがありましたが、
PCを再起動すると再度走行できるようになりました。

当方案件はCloseで構いません。

@kosei1515 さま　@kasecato さま
当方の案件は解決しましたので、案件Closeが最後の方がISSUEをCloseしていただきますよう宜しくお願い致します。

kasecato, 2023-01-15:
@hidechun san
`1/14日版のDockerイメージ（tar版）` は [README.md](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim#%E4%BA%8B%E5%89%8D%E6%BA%96%E5%82%99-1) にあるリンクでしょうか？ 
https://drive.google.com/file/d/145HyoeXye_bbdT6tOVVCvbSM1MTm2CKI/view?usp=share_link

docker のi mage id は `41262e474b24` でしょうか？自分はこのバージョンでローカルでは動いていません。
```bash
$ docker images | grep ai 
ghcr.io/automotiveaichallenge/aichallenge2022-sim/autoware-universe-cuda   3.1                           41262e474b24   9 days ago     22.7GB
ghcr.io/automotiveaichallenge/aichallenge2022-sim/autoware-universe-cuda   latest                        ab92f06f6314   5 weeks ago    19.8GB

```

hidechun, 2023-01-15:
@kasecato さま
添付のリンクで合っています。DL後、ファイルをhomeに配置し、以下コマンドで取り込みました。
docker load < aichallenge2022_sim_autoware_v3.1.tar.gz

＞docker の image id は 41262e474b24 でしょうか？ 
私のPC環境は以下です。
ーーーーーーーーーーーーーー
$ docker images | grep ai
aichallenge-eval                                                           latest                        0316cf2239ce   24 hours ago    23GB
ghcr.io/automotiveaichallenge/aichallenge2022-sim/autoware-universe-cuda   3.1                           41262e474b24   9 days ago      22.7GB
ghcr.io/automotiveaichallenge/aichallenge2022-sim/autoware-universe-cuda   latest                        ab92f06f6314   5 weeks ago     19.8GB
ghcr.io/automotiveaichallenge/aichallenge2022-sim/autoware-universe-cuda   <none>                        3cae0851bbf1   2 months ago    19.7GB
ーーーーーーーーーーーーーー

＞また，走行できない状態があるのは 10 回起動して何回ぐらいのイメージでしょうか？
説明文での連絡となって申し訳ありません。
　・Reee009876　さまからのAWSIM同時起動の連絡を受け、
　　PC起動と同時にローカル環境でAWSIM起動後にrun.shでローカルシミュレーション実施＞＞動作OK
　・AWSIMやターミナルをすべてCloseして再度シミュレーション実施
　　　・５分程度待っても走行しない
　　　・２分くらい経過後に走行を開始したが、最初の交差点を右折する前で走行と停止を繰り返すのでAWSIMとターミナルを強制クローズ
　　　・AWSIMを立ち上げると、立ち上げと同時に左折ウインカが点灯しており動作がおかしくなってきた（5回立ち上げなおすも改善せず）
　・PCを再起動し、再度ローカルシミュレーションを実施＞＞動作OK

hidechun, 2023-01-15:
上記コメント
＞・PCを再起動し、再度ローカルシミュレーションを実施＞＞動作OK
Logを添付致します。参考になれば幸いです。
[challange_log_Local2.txt](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/files/10419658/challange_log_Local2.txt)

kasecato, 2023-01-15:
`ndt_scan_matcher` が動きが違うようですが，3.1 で何か変更があったのでしょうか。素人の自分ではこれ以上はわからないです 😿

- hidechun san のログ
```bash
[pose_initializer_node-17] [INFO] [1673766452.090497282] [localization.util.pose_initializer_node]: Call NDT align server.
[pose_initializer_node-17] [INFO] [1673766452.964652943] [localization.util.pose_initializer_node]: NDT align server succeeded.
[pose_initializer_node-17] [INFO] [1673766452.964803326] [localization.util.pose_initializer_node]: service exit: /localization/initialize
[pose_initializer_node-17] status:
[pose_initializer_node-17]   success: true
[pose_initializer_node-17]   code: 0
[pose_initializer_node-17]   message: ""
[pose_initializer_node-17] 
```

- 私のログ
```bash
[pose_initializer_node-17] [INFO] [1673768562.403102258] [localization.util.pose_initializer_node]: Call NDT align server.
[pose_initializer_node-17] [INFO] [1673768562.403497901] [localization.util.pose_initializer_node]: NDT align server failed.
[pose_initializer_node-17] [INFO] [1673768562.403583332] [localization.util.pose_initializer_node]: service exit: /localization/initialize
[pose_initializer_node-17] status:
[pose_initializer_node-17]   success: false
[pose_initializer_node-17]   code: 4
[pose_initializer_node-17]   message: "NDT align server failed."
[pose_initializer_node-17] 
```

kasecato, 2023-01-15:
@hidechun san
すみません，自分は CUDA が原因でした。nvidia driver をアップデートしたことで CUDA が動かなくなったようです。ロールバックしてNvidia Container Toolkit を再インストールしたところ解決しました。
```bash
[component_container_mt-38] [TRT_COMMON][WARNING]: CUDA initialization failure with error: 999. 
```

hidechun, 2023-01-15:
@kasecato さま　動作されたようで良かったです！
@kosei1515 さま　解決されたら、Closeをお願い致します。

hidechun, 2023-01-15:
動作せず状態が再現しましたのでLog取得しました。
参考情報として添付しておきます。
[challange_log_NG.txt](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/files/10419764/challange_log_NG.txt)

[sample-65] Planning.....
から先に進めないようです。AWSIM上の車の灯火器（ブレーキランプ・ウィンカ）は消灯したままです。

kosei1515, 2023-01-15:
@hidechun 様

ありがとうございます
1/14日付のもので試しましたが，上記同様のエラーでシミュレーションがスタートしませんでした

試しにローカル環境で評価なしに実行してみたところ，
Awsimを起動してからAutowareを起動するとシミュレーションがスタートせず，Autowareを起動してからAWSIMを起動するとシミュレーションがスタートしました

同様のエラーは見られていないでしょうか？

ちなみにスタートしない際のエラーとして散見されるのは以下です

```
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[component_container_mt-3] [ERROR] [1673780505.235844528] [system.system_monitor.hdd_monitor]: socket connect error. Connection refused
[component_container_mt-3] [ERROR] [1673780505.235893369] [system.system_monitor.hdd_monitor]: Failed to unmount device : /dev/sda3[/home/kosei/AI_Challenge_2022_winter/aichallenge2022-sim/autoware]
[sample-71] Planning.....
[sample-71] Planning.....
```

kosei1515, 2023-01-15:
@hidechun 
追加の知見として，初期位置を違う場所にはめてから再度もとの位置にはめ直すとシミュレーションがスタートしました

hidechun, 2023-01-15:
@kosei1515 さま
このトピックは【オンライン評価環境にソースコードを提出する際の手順におけるローカル評価環境でのスコア算出】での
aichallenge_submit.tar.gzを使った動作に関する問い合わせでしたので、シミュレータ環境下での確認はしておりません。

＞Awsimを起動してからAutowareを起動するとシミュレーションがスタートせず，Autowareを起動してからAWSIMを起動するとシミュレーションがスタートしました
シミュレータ環境下での確認は常にAWSIMを先に立ち上げてからシミュレーションしておりましたので、
逆の状態での確認は出来ておりません。
同様の事象が発生したらLog取りやRViz上の自車位置修正など試してみようと思います。

hidechun, 2023-01-22:
@Reee009876 さま　@kosei1515 さま
本issueでの問い合わせ内容についてはすでにClose、
解析中に発見された別問題は関連トピックについては #26 #27 を起票頂いているので、
こちらはCloseしておきます。

closed 2023-01-22 by hidechun