AutomotiveAIChallenge/aichallenge2022-sim #22: パラメータが表示されない件について

Kenzo-Fujisaki, 2023-01-13:
### 環境
* OS: Ubuntu 20.04 LTS
* CPU: AMD Ryzen 7 5800X 8-Core Processor(8 cores)
* GPU:  NVIDIA Geforce RTX 3090(VRAM 24 GB)
* RAM: 64GB
* ROM: SSD 2.0TB
* ROS2: galactic

### 実行手順
1. AWSIM起動
2. `ros2 launch aichallenge_launch aichallenge.launch.xml`
3. `ros2 param list`

### 問題
`ros2 param list`を実行した際にdaemonがクラッシュしたというレポートが現れ、パラメータが見れない状況です。対処法はありますでしょうか。`ros2 topic list`は問題なく動きました。

[image]

kosei1515, 2023-01-15:
@Kenzo-Fujisaki 様
参加者のものですが
こちら当方の環境でも同様の現象を確認しています．
現状の打開策にはなりますがこちらでの対応法を共有させていただきます

こちらではrqtのDynamic Reconfigureをつかってパラメータを確認しています

コンソールで`source`した後に
```
rqt
```
と入力するとUIが出現します
`Plugin`タブから`Configuration`，`Dynamic Reconfigure`を選択するとさらにプラグインが表示され，Filter keyの部分でキーワード検索（例えばplanningとするとplanning系のノード群が出てきます　`ros2 node list`で調べたいノードのノード名をみてそれを参考に検索するといいと思います）するとノードごとのパラメータを確認できます

こちら参考サイトです

http://wiki.ros.org/rqt_reconfigure

Kenzo-Fujisaki, 2023-01-16:
@kosei1515 様

上記の方法で無事パラメータを確認することが出来ました。
ご教授頂きありがとうございます。

Reee009876, 2023-01-16:
お世話になっております。

こちらの環境ではクラッシュはしないものの、10分以上待っても表示されないことを確認しております（時間がかかっているのかも確認中）。
※ノード指定をした場合は表示されます、

対処法・原因がわかるまでは
申し訳ありませんが、kosei1515様にご提案いただいた方法かノード指定でのparam表示で、ご対処いただいてもよろしいでしょうか？

Kenzo-Fujisaki, 2023-01-17:
承知しました。
ご対応頂きありがとうございます。

closed 2023-01-25 by Kenzo-Fujisaki