AutomotiveAIChallenge/aichallenge2022-sim #24: オンライン環境のアップロード制限が無効

kasecato, 2023-01-14:
オンライン環境の以下の 2 つの制限事項が現在無効のようです。

> なお、採点実行中は新たなソースのアップロードはできません。またアップロードできるのは1日3回までで、日本時間0時にリセットされます。

1. 採点実行中は新たなソースのアップロードはできません
2023-01-13 22:32 | 採点完了 | x秒
2023-01-13 22:32 | 採点完了 | x秒
1. アップロードできるのは1日3回までで、日本時間0時にリセット
2023-01-13 22:32 | 採点完了 | x秒
2023-01-13 22:32 | 採点完了 | x秒
2023-01-13 20:07 | 結果なし |  
2023-01-13 18:50 | 結果なし |  
2023-01-13 17:55 | 採点完了 | x秒
2023-01-13 17:00 | 採点完了 | x秒
2023-01-13 16:08 | 採点完了 | x秒
2023-01-13 15:45 | ビルドエラー |  
2023-01-13 15:11 | 採点完了 | x秒
2023-01-13 11:11 | 結果なし |  
2023-01-13 09:51 | 採点完了 | x秒
2023-01-13 01:06 | 結果なし

- README.md, 「[(3) オンライン評価環境webページよりソースをアップロードする](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/main/README.md#3-%E3%82%AA%E3%83%B3%E3%83%A9%E3%82%A4%E3%83%B3%E8%A9%95%E4%BE%A1%E7%92%B0%E5%A2%83web%E3%83%9A%E3%83%BC%E3%82%B8%E3%82%88%E3%82%8A%E3%82%BD%E3%83%BC%E3%82%B9%E3%82%92%E3%82%A2%E3%83%83%E3%83%97%E3%83%AD%E3%83%BC%E3%83%89%E3%81%99%E3%82%8B)」

Reee009876, 2023-01-15:
お世話になっております。

明日、オンライン環境の確認を行います為、もうしばらくお待ちください。

Reee009876, 2023-01-17:
こちら、修正が完了いたしました。

closed 2023-01-17 by kasecato

kasecato, 2023-01-17:
アップロードし過ぎた日があったため，総量を超えないように自粛します🙇‍♂️ 3回*25日間=75回

Reee009876, 2023-01-17:
こちらの不備にも関わらず、ご配慮いただき、ありがとうございます。