AutomotiveAIChallenge/aichallenge2022-sim #26: AWSIM起動後にAutowareを起動し，シミュレーションがすぐにスタートする際のエラーについて

kosei1515, 2023-01-15:
表題の件に関して，https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/issues/20　に関連して報告させていただきます

数回AWSIMとAutowareの起動を試していると，AWSIM起動後にAutowareを起動し，シミュレーションがすぐにスタートする場合があります

その際，添付の動画のようにPerceptionの起動が著しく遅いです

オンラインにおける採点時に問題になりうると考えたため，報告させていただきます

[Screencast from 01-15-2023 08:23:06 PM.webm](https://user-images.githubusercontent.com/81236654/212538220-bd072752-b88b-432c-b980-c0570edb8be0.webm)

Reee009876, 2023-01-17:
お世話になっております．

シミュレーションがすぐにスタートするということですが，問題ないように見受けられたのですが，もう少し詳細をお願いできないでしょうか？

また，perception の起動が遅いとのことですが，ノードの立ち上がりが遅いのか，認識が遅いのかどちらになりますか？？

五月雨で申し訳ありませんが，ご協力のほどよろしくお願いします．

kosei1515, 2023-01-17:
> シミュレーションがすぐにスタートするということですが，問題ないように見受けられたのですが，もう少し詳細をお願いできないでしょうか？

perceptionのノードがはじめからきちんと起動するのであれば問題ありません

> ノードの立ち上がりが遅いのか，認識が遅いのかどちらになりますか？？
こちらは調べていません
ただ，こちらで調べたところ，center_pointというライダーベースの認識に関するノードからPublishされる認識結果(`Detected Object`)の初回のPublishまでに時間がかかっているようです

起動部分はこちらです
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/dd263d3057f658af1c387d96392fd7bc1006457c/autoware/aichallenge_ws/src/aichallenge_submit/tier4_perception_launch/launch/object_recognition/detection/camera_lidar_fusion_based_detection.launch.xml#L159

kosei1515, 2023-01-17:
こちらに関してさらに知見として，
`sample_code_cpp`におけるすべてのトピックのPublishをある程度遅らせることで，状況が改善しています

例としてこちらでは
[コンストラクタの最後の部分](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/dd263d3057f658af1c387d96392fd7bc1006457c/autoware/aichallenge_ws/src/aichallenge_submit/sample_code_cpp/src/sample.cpp#L42)に1分間待機するプログラムを導入することで正常に動作しています

```
# example
using namespace std::chrono_literals;
rclcpp::sleep_for(60000ms);
```

Reee009876, 2023-01-17:
情報提供ありがとうございます。

現在、詳細確認かつ対応の検討中であるため、もうしばらくお時間をいただきたいです。

Reee009876, 2023-01-24:
お世話になっております。

根本的な解決法の調査に時間を要しております。
判明するまで、こちらの方法でご対処いただけないでしょうか？
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/issues/26#issuecomment-1384788793

よろしくおねがいします。

kosei1515, 2023-01-25:
わかりました
確認ですが上記のような実装でも計測タイムに影響はないでしょうか？

dazzhe, 2023-01-27:
@kosei1515 
返信遅くなり申し訳ございません。
計測は車両が動き出したタイミングで開始されるため、計測タイムに影響はございません。

closed 2023-01-27 by kosei1515

hidechun, 2023-01-28:
当方も本トピックの1分待機を入れることで、シミュレーションしない事象が解決したようです。
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/issues/20#issuecomment-1383092438

有用な情報共有ありがとうございます。