AutomotiveAIChallenge/aichallenge2022-sim #32: planningが終わらない現象について

moritakko, 2023-01-23:
手順通りシミュレーションを実行しようとすると、
２回に１回くらいの頻度でplanningが終わらずシミュレーションが開始しません。

https://drive.google.com/file/d/1RZkiEFu5L8dlcTmRLI_SW_Uq2nVdEoP8/view?usp=sharing

```bash
$ source install/setup.bash
$ ros2 launch aichallenge_launch aichallenge.launch.xml
```

moritakko, 2023-01-23:
logの一部です（ずっとplanningの際のログです）
```
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[rviz2-66] [INFO] [1674462411.025363508] [rviz2]: Message Filter dropping message: frame 'base_link' at time 9.290 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sample-65] Planning.....
[component_container_mt-4] [ERROR] [1674462411.138859945] [system.system_monitor.hdd_monitor]: Failed to execute findmnt. /dev/sda3
[sample-65] Planning.....
[rviz2-66] [INFO] [1674462411.220045568] [rviz2]: Message Filter dropping message: frame 'base_link' at time 9.490 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_mt-47] [WARN] [1674462411.228500319] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: waiting for perception object ...
[sample-65] Planning.....
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[rviz2-66] [INFO] [1674462411.431318441] [rviz2]: Message Filter dropping message: frame 'base_link' at time 9.690 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sample-65] Planning.....
[component_container_mt-47] [WARN] [1674462411.584289937] [planning.scenario_planning.lane_driving.motion_planning.surround_obstacle_checker]: waiting for dynamic object info...
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[component_container_mt-47] [WARN] [1674462411.612983295] [util]: [Avoidance] MPT:  OSQP solution status: OSQP_SOLVED_INACCURATE
[sample-65] Planning.....
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[sample-65] Planning.....
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[sample-65] Planning.....
[component_container_mt-4] [ERROR] [1674462412.138134885] [system.system_monitor.hdd_monitor]: Failed to execute findmnt. /dev/sda3
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[component_container-50] [INFO] [1674462412.234715668] [control.trajectory_follower.lane_departure_checker_node]: waiting for reference_trajectory msg...
[sample-65] Planning.....
[sample-65] Planning.....
[sample-65] Planning.....
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[system_error_monitor-7] [WARN] [1674462412.633382332] [system_error_monitor /autoware/sensing/node_alive_monitoring]: [Latent Fault]: Warning
[system_error_monitor-7] [WARN] [1674462412.633424475] [system_error_monitor /autoware/sensing/node_alive_monitoring/topic_status/ad_service_state_monitor: sensing_topic_status]: [Latent Fault]: Warn
[system_error_monitor-7] [ERROR] [1674462412.633431775] [system_error_monitor /autoware/control/autonomous_driving/node_alive_monitoring]: [Single Point Fault]: Error
[system_error_monitor-7] [ERROR] [1674462412.633439530] [system_error_monitor /autoware/control/autonomous_driving/node_alive_monitoring/topic_status/ad_service_state_monitor: control_topic_status]: [Single Point Fault]: Error
[system_error_monitor-7] [ERROR] [1674462412.633444326] [system_error_monitor /autoware/perception/node_alive_monitoring]: [Single Point Fault]: Error
[system_error_monitor-7] [ERROR] [1674462412.633448991] [system_error_monitor /autoware/perception/node_alive_monitoring/topic_status/ad_service_state_monitor: perception_topic_status]: [Single Point Fault]: Error
[system_error_monitor-7] [ERROR] [1674462412.633453674] [system_error_monitor /autoware/planning/node_alive_monitoring]: [Single Point Fault]: Error
[system_error_monitor-7] [ERROR] [1674462412.633458144] [system_error_monitor /autoware/planning/node_alive_monitoring/topic_status/ad_service_state_monitor: planning_topic_status]: [Single Point Fault]: Error
[component_container_mt-47] [WARN] [1674462412.639685563] [util]: [Avoidance] MPT:  OSQP solution status: OSQP_SOLVED_INACCURATE
[sample-65] Planning.....
[component_container-50] [WARN] [1674462412.766516253] [control.operation_mode_transition_manager]: Engage unavailable: trajectory size must be > 2
[sample-65] Planning.....
[sample-65] Planning.....
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[sample-65] Planning.....
[component_container_mt-4] [ERROR] [1674462413.136679791] [system.system_monitor.hdd_monitor]: Failed to execute findmnt. /dev/sda3
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[sample-65] Planning.....
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[... cut ...]
[component_container_mt-4] [ERROR] [1674462421.138987176] [system.system_monitor.hdd_monitor]: Failed to execute findmnt. /dev/sda3
[sample-65] Planning.....
[component_container-26] [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[sample-65] Planning.....
[component_container_mt-47] [WARN] [1674462421.338144157] [util]: [Avoidance] MPT:  OSQP solution status: OSQP_SOLVED_INACCURATE
[component_container_mt-47] [WARN] [1674462421.350122101] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: waiting for perception object ...
[sample-65] Planning.....
```

Reee009876, 2023-01-24:
お世話になっております。

解決策が判明するまで、こちらの方法でご対処願えないでしょうか？？https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/issues/20#issuecomment-1383122492

moritakko, 2023-01-26:
追記です。

提出前の動作確認の際に、evaluationディレクトリで
```bash
$ sh create_submit_tar.sh 
$ sh build.sh Challenge
$ sh run.sh
```
した際には、この現象は確認できないです。

Reee009876, 2023-01-27:
申し訳ございません。

現在、事象の原因や
どのような操作を行えば発生するのかの特定もいたっていない状況です。

大変お手数ですが、
[こちら](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/issues/32#issuecomment-1401752333)で
ご対応いただければ幸いです。

kasecato, 2023-01-27:
こちらで提案されている `sample_code.cpp` のコンストラクタでスリープを 1 分間使用するワークアラウンドではいかがでしょうか？
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/issues/26#issuecomment-1384788793