AutomotiveAIChallenge/aichallenge2022-sim #34: 各イベントについて記載している走破方法以外の経路・方法で走破した場合の採点が成功したときについて

kasecato, 2023-01-25:
RULE.md の[「シナリオ」](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/release_20221229/RULE.md#%E3%82%B7%E3%83%8A%E3%83%AA%E3%82%AA)に以下の記載があります。

> 各イベントについて記載している走破方法以外の経路・方法で走破した場合、オンライン採点環境では動作しない可能性があります。

オンライン採点環境チャレンジ コースの「低速車両への追従」のシナリオにつきまして，もし「追従ではなく車両を追い越した」場合でも採点ステータスが「採点完了」と評価されれば，有効なスコアとなりますでしょうか？または最終判定で失格となりますでしょうか？

- RULE.md, 「シナリオ」, https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/blob/release_20221229/RULE.md#%E3%82%B7%E3%83%8A%E3%83%AA%E3%82%AA

hir-yk, 2023-01-25:
のっかります。運営の方にご回答いただければと思います。 今回のイベントは競技として「ぶつからず早くゴールする」を優先として良いでしょうか？  本issueの「左から追い越す」は実路ではNGな感じですが本競技でがアリですか？

Reee009876, 2023-01-25:
お世話になっております。

記載している走破方法以外の方法を用いて走破した場合のスコアは無効となります。
そのため、「低速車両への追従」において”左から追い越す”はNGです。

kasecato, 2023-01-25:
@Reee009876 san
その場合に採点結果が「結果なし」となる認識でよろしいでしょうか？

Reee009876, 2023-01-26:
申し訳ございません。
前回のご案内に一部誤りがありました。

RULE.mdに書かれていない方法で走破した場合でも
オンライン環境において採点完了と表示されれば、有効なスコアとしてみなされます。

しかし、オンライン採点環境では障害物となる車両等が追加されているため、
基本的にはRULE.mdで記載されている方法以外では走破できないようになっています。

kasecato, 2023-01-26:
@Reee009876 san
オンライン環境において採点完了となれば有効なスコアとなる旨把握いたしました。
ご確認ありがとうございました :bow:

closed 2023-01-26 by kasecato