AutomotiveAIChallenge/aichallenge2022-sim #35: dockerイメージの入手方法について

kbys0519, 2023-01-26:
READMEに従って環境構築をしているのですが、Dockerイメージの入手時またはrockerの起動時にエラーまたはスタックが発生します。
docker, rocker, nvidia container toolkit, git lfs, ros galacticの環境構築はすで完了しています。
`docker pull ghcr.io/automotiveaichallenge/aichallenge2022-sim/autoware-universe-cuda:3.1`
を実行するとダウンロードに時間がかかり、タイムアウトになってしまいます。
そのため指示通りGoogleDriveの方からdockerイメージをダウンロードし、ロードはできました。
しかし、下記コマンドを実行するとエラーが出るわけでもなく画像の状態でスタックしてしまい、この先に進めません。
`rocker --nvidia --x11 --user --net host --privileged --volume autoware:/aichallenge -- ghcr.io/automotiveaichallenge/aichallenge2022-sim/autoware-universe-cuda:3.1`
[image]

Reee009876, 2023-01-26:
お世話になっております。

一旦、PCを再起動して
再度実行しても同じ状況でしょうか？

kbys0519, 2023-01-26:
お世話になっております．再起動や複数のPCで試しても同じ状況でした．

Reee009876, 2023-01-27:
申し訳ございません。
こちらの環境では再現できていないため、
解決策のご提案が今すぐにさせていただくのは難しい状況です。

その状態になってからどれぐらいお待ちになられてますか？？
また、該当ターミナルは固まるのでしょうか？それとも、進まないだけでしょうか？

kbys0519, 2023-01-29:
半日以上待っても変化なしでした。docker pullでdocker imageのダウンロードを複数回試みたところタイムアウトにならずダウンロードでき、無事Autowareを起動することができました。Google Driveに上がっている方のdockert imageがうまく動作しなかったのかと思います。お手数をおかけしました。ありがとうございました。

MizuhoAOKI, 2023-02-06:
@kbys0519 
今更ですが, 同様の件を経験しましたので知見を共有いたします. 
Google Driveからダウンロードしたdocker imageの中身自体に問題はないのですが, docker loadしただけではイメージに適切な名前がついていないのが問題です. 

`sudo docker tag イメージID ghcr.io/automotiveaichallenge/aichallenge2022-sim/autoware-universe-cuda:3.1`

このように名前を設定した後に`rocker`コマンドを実行すれば, 適切に動作しました.

(イメージに名前がついていないと, rocker起動時にghcr.io...という名のイメージを探すもローカルに存在しない扱いになるため, オンラインから重いデータをpullし始め, 外から見ると処理が止まっているように見えるのだと思います.)