AutomotiveAIChallenge/aichallenge2022-sim #36: 動作確認の質問について

Phyrisana, 2023-01-27:
動作確認
AutowareとAWSIMを用いて動作確認を行う方法を記載します。

AWSIMを起動
Autowareを起動

# Rockerコンテナ内で
cd /aichallenge

ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=nishishinjuku_autoware_mapを入力すると、

```
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-01-27-17-07-18-576097-phyrisana-OMEN-by-HP-Laptop-16-b0xxx-189
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=PackageNotFoundError("package 'tier4_vehicle_launch' not found, searching: ['/autoware/install/autoware_launch', '/autoware/install/web_controller', '/autoware/install/vls_description', '/autoware/install/sample_sensor_kit_launch', '/autoware/install/awsim_sensor_kit_launch', '/autoware/install/common_sensor_launch', '/autoware/install/common_awsim_sensor_launch', '/autoware/install/velodyne_pointcloud', '/autoware/install/velodyne_driver', '/autoware/install/velodyne_msgs', '/autoware/install/velodyne_monitor', '/autoware/install/sample_sensor_kit_description', '/autoware/install/awsim_sensor_kit_description', '/autoware/install/velodyne_description', '/autoware/install/vehicle_velocity_converter', '/autoware/install/vehicle_cmd_gate', '/autoware/install/trajectory_follower_nodes', '/autoware/install/pure_pursuit', '/autoware/install/trajectory_follower', '/autoware/install/tier4_planning_rviz_plugin', '/autoware/install/surround_obstacle_checker', '/autoware/install/steer_offset_estimator', '/autoware/install/simple_planning_simulator', '/autoware/install/pacmod_interface', '/autoware/install/operation_mode_transition_manager', '/autoware/install/obstacle_stop_planner', '/autoware/install/obstacle_cruise_planner', '/autoware/install/obstacle_collision_checker', '/autoware/install/obstacle_avoidance_planner', '/autoware/install/behavior_velocity_planner', '/autoware/install/motion_velocity_smoother', '/autoware/install/behavior_path_planner', '/autoware/install/lane_departure_checker', '/autoware/install/ground_segmentation', '/autoware/install/control_performance_analysis', '/autoware/install/global_parameter_loader', '/autoware/install/freespace_planner', '/autoware/install/vehicle_info_util', '/autoware/install/vehicle_constants_manager', '/autoware/install/twist2accel', '/autoware/install/lidar_apollo_segmentation_tvm_nodes', '/autoware/install/lidar_apollo_segmentation_tvm', '/autoware/install/tvm_utility', '/autoware/install/trtexec_vendor', '/autoware/install/traffic_light_visualization', '/autoware/install/traffic_light_ssd_fine_detector', '/autoware/install/traffic_light_map_based_detector', '/autoware/install/traffic_light_classifier', '/autoware/install/topic_state_monitor', '/autoware/install/motion_common', '/autoware/install/time_utils', '/autoware/install/tier4_vehicle_rviz_plugin', '/autoware/install/tier4_calibration_rviz_plugin', '/autoware/install/external_cmd_selector', '/autoware/install/awapi_awiv_adapter', '/autoware/install/autoware_iv_internal_api_adaptor', '/autoware/install/autoware_iv_external_api_adaptor', '/autoware/install/tier4_auto_msgs_converter', '/autoware/install/external_cmd_converter', '/autoware/install/accel_brake_map_calibrator', '/autoware/install/raw_vehicle_cmd_converter', '/autoware/install/tier4_vehicle_msgs', '/autoware/install/dummy_infrastructure', '/autoware/install/tier4_v2x_msgs', '/autoware/install/tier4_traffic_light_rviz_plugin', '/autoware/install/tier4_api_msgs', '/autoware/install/tier4_system_msgs', '/autoware/install/tier4_state_rviz_plugin', '/autoware/install/fault_injection', '/autoware/install/tier4_simulation_msgs', '/autoware/install/tier4_simulated_clock_rviz_plugin', '/autoware/install/tier4_screen_capture_rviz_plugin', '/autoware/install/rtc_replayer', '/autoware/install/rtc_manager_rviz_plugin', '/autoware/install/rtc_interface', '/autoware/install/rtc_auto_mode_manager', '/autoware/install/tier4_rtc_msgs', '/autoware/install/scenario_selector', '/autoware/install/planning_debug_tools', '/autoware/install/external_velocity_limit_selector', '/autoware/install/costmap_generator', '/autoware/install/tier4_planning_msgs', '/autoware/install/tier4_perception_rviz_plugin', '/autoware/install/tensorrt_yolo', '/autoware/install/detection_by_tracker', '/autoware/install/shape_estimation', '/autoware/install/object_merger', '/autoware/install/multi_object_tracker', '/autoware/install/lidar_apollo_instance_segmentation', '/autoware/install/image_projection_based_fusion', '/autoware/install/euclidean_cluster', '/autoware/install/dummy_perception_publisher', '/autoware/install/detected_object_validation', '/autoware/install/detected_object_feature_remover', '/autoware/install/ad_service_state_monitor', '/autoware/install/tier4_perception_msgs', '/autoware/install/occupancy_grid_map_outlier_filter', '/autoware/install/compare_map_segmentation', '/autoware/install/pointcloud_preprocessor', '/autoware/install/tier4_pcl_extensions', '/autoware/install/tier4_localization_rviz_plugin', '/autoware/install/pose_initializer', '/autoware/install/ndt_scan_matcher', '/autoware/install/initial_pose_button_panel', '/autoware/install/ad_api_adaptors', '/autoware/install/tier4_localization_msgs', '/autoware/install/tier4_hmi_msgs', '/autoware/install/tier4_control_rviz_plugin', '/autoware/install/joy_controller', '/autoware/install/tier4_api_utils', '/autoware/install/system_monitor', '/autoware/install/elevation_map_loader', '/autoware/install/autoware_external_api_msgs', '/autoware/install/tier4_external_api_msgs', '/autoware/install/tier4_debug_tools', '/autoware/install/system_error_monitor', '/autoware/install/mission_planner', '/autoware/install/route_handler', '/autoware/install/radar_tracks_msgs_converter', '/autoware/install/radar_fusion_to_detected_object', '/autoware/install/probabilistic_occupancy_grid_map', '/autoware/install/planning_evaluator', '/autoware/install/planning_error_monitor', '/autoware/install/lidar_centerpoint', '/autoware/install/perception_utils', '/autoware/install/path_distance_calculator', '/autoware/install/map_based_prediction', '/autoware/install/motion_utils', '/autoware/install/map_loader', '/autoware/install/localization_evaluator', '/autoware/install/kinematic_evaluator', '/autoware/install/interpolation', '/autoware/install/imu_corrector', '/autoware/install/image_diagnostics', '/autoware/install/heatmap_visualizer', '/autoware/install/grid_map_utils', '/autoware/install/goal_distance_calculator', '/autoware/install/front_vehicle_velocity_estimator', '/autoware/install/freespace_planning_algorithms', '/autoware/install/emergency_handler', '/autoware/install/ekf_localizer', '/autoware/install/dummy_diag_publisher', '/autoware/install/crosswalk_traffic_light_estimator', '/autoware/install/tier4_autoware_utils', '/autoware/install/stop_filter', '/autoware/install/pose2twist', '/autoware/install/gnss_poser', '/autoware/install/tier4_debug_msgs', '/autoware/install/tier4_datetime_rviz_plugin', '/autoware/install/tier4_control_msgs', '/autoware/install/tamagawa_imu_driver', '/autoware/install/simulator_compatibility_test', '/autoware/install/signal_processing', '/autoware/install/shift_decider', '/autoware/install/sample_vehicle_launch', '/autoware/install/sample_vehicle_description', '/autoware/install/polar_grid', '/autoware/install/pointcloud_to_laserscan', '/autoware/install/pandar_description', '/autoware/install/osqp_interface', '/autoware/install/object_range_splitter', '/autoware/install/neural_networks_provider', '/autoware/install/ndt', '/autoware/install/ndt_pcl_modified', '/autoware/install/ndt_omp', '/autoware/install/mussp', '/autoware/install/motion_testing', '/autoware/install/morai_msgs', '/autoware/install/map_tf_generator', '/autoware/install/localization_error_monitor', '/autoware/install/livox_tag_filter', '/autoware/install/livox_description', '/autoware/install/lanelet2_map_preprocessor', '/autoware/install/lanelet2_extension', '/autoware/install/kalman_filter', '/autoware/install/individual_params', '/autoware/install/imu_description', '/autoware/install/image_transport_decompressor', '/autoware/install/gyro_odometer', '/autoware/install/grid_map', '/autoware/install/grid_map_demos', '/autoware/install/grid_map_visualization', '/autoware/install/grid_map_sdf', '/autoware/install/grid_m [... cut ...] autoware_auto_planning_msgs', '/autoware/install/autoware_auto_perception_rviz_plugin', '/autoware/install/autoware_auto_perception_msgs', '/autoware/install/autoware_auto_mapping_msgs', '/autoware/install/autoware_auto_geometry_msgs', '/autoware/install/autoware_auto_control_msgs', '/autoware/install/autoware_auto_common', '/autoware/install/autoware_adapi_version_msgs', '/autoware/install/autoware_adapi_v1_msgs', '/autoware/install/autoware_ad_api_specs', '/opt/ros/galactic']"

```
こいう異常がでました。

kasecato, 2023-01-27:
`rocker` で docker 起動する毎に ROS2 をセットアップするファイルを `source` コマンドで実行する必要があると思われます。

```bash
# ビルドする
cd /aichallenge/aichallenge_ws
rosdep update
rosdep install -y -r -i --from-paths src --ignore-src --rosdistro $ROS_DISTRO
colcon build

# ROS2 をファイルでセットアップする ↓↓↓↓このステップが必要と思われます↓↓↓↓
source /aichallenge/aichallenge_ws/install/setup.bash

cd /aichallenge
ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=nishishinjuku_autoware_map
```

- https://github.com/AutomotiveAIChallenge/aichallenge2022-sim#%E3%82%B5%E3%83%B3%E3%83%97%E3%83%AB%E3%82%B3%E3%83%BC%E3%83%89%E8%B5%B7%E5%8B%95
- https://docs.ros.org/en/foxy/Tutorials/Beginner-CLI-Tools/Configuring-ROS2-Environment.html#source-the-setup-files

Phyrisana, 2023-01-27:
> `rocker`de docker 启动するるニ ROS2 をズットアップするファイルを`source`コマンド実行する必须要ガあると思わレま。
> 
> ```shell
> # Step 1: ビルドする
> cd /aichallenge/aichallenge_ws
> rosdep update
> rosdep install -y -r -i --from-paths src --ignore-src --rosdistro $ROS_DISTRO
> colcon build
> 
> # Step 2: ROS2 をファイルでセットアップする ↓↓↓↓このステップが必要と思われます↓↓↓↓
> source /aichallenge/aichallenge_ws/install/setup.bash
> 
> # Step 3: 動作確認
> cd /aichallenge
> ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=nishishinjuku_autoware_map
> ```
> 
> * [https://github.com/AutomotiveAIChallenge/aichallenge2022-sim#%E3%82%B5%E3%83%B3%E3%83%97%E3%83%AB%E3%82%B3%E3%83%BC %E3%83%89%E8%B5%B7%E5%8B%95](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim#%E3%82%B5%E3%83%B3%E3%83%97%E3%83%AB%E3%82%B3%E3%83%BC%E3%83%89%E8%B5%B7%E5%8B%95)
> * https://docs.ros.org/en/foxy/Tutorials/Beginner-CLI-Tools/Configuring-ROS2-Environment.html#source-the-setup-files

ありがとうございます。解決できましたが、AWSIMとAutowareを一緒起動する際には強制再起動されます。
osと関係ありますでしょうか。こちらが使っているOSはUbuntu 22.04

kasecato, 2023-01-27:
2つ可能性があると思います。

1. AWSIM が起動できずクラッシュする場合
Ubuntu にインストールされた NVIDIA ドライバが原因かもしれないです。以下の記事からドライバを更新すると AWSIM がクラッシュしなくなる可能性があります。
https://twitter.com/wd01x2oa/status/1616420832401817601?s=46&t=3a0mNaOhKySYOvZXU8CuHw

2. AWSIM と Autoware が連携しない場合
今回のコンテナ側の Autoware は ROS2 Galactic を使用しているため，Galactic が対応している Ubuntu 20.04 が必要かもしれないです。ホスト側を Ubuntu 20.04 にして ROS2 Galactic をインストールします。

Reee009876, 2023-01-27:
[kasecato](https://github.com/kasecato)さま
情報の共有ありがとうございます。

[kasecato](https://github.com/kasecato)さま
kasecatoさまがおっしゃっていることに加えて、
[READ.ME](https://github.com/AutomotiveAIChallenge/aichallenge2022-sim#%E5%8B%95%E4%BD%9C%E7%92%B0%E5%A2%83)記載のスペック以下だと、動作しない・不安定であることをこちらでは確認しておりますため、ご確認いただけますか？

kasecato, 2023-01-27:
スペックもあると思います。実行時に free -h コマンドで swap が一杯になっているとメモリ不足で OS が不安定になり AWSIM もクラッシュします。AWSIM と Autoware を 1台で起動する場合，メモリは swap と合わせて 20GB は必要な印象です。

closed 2023-01-29 by Phyrisana