AutomotiveAIChallenge/aichallenge2022-sim #37: チャレンジコースの低速車両追従シナリオはローカルとオンラインで動作の差分が大きくありますか？

kasecato, 2023-01-28:
チャレンジ コースの低速車両の追従につきまして，ローカル環境では以下の図のように低速車両が右ウインカで車線変更して自車両の前に障害物としてカットインしてくると思います。

この低速車両が車線変更で自車両の前にカットインする動作はオンライン環境でも同じでしょうか？

[image]

質問の背景としましては，もし低速車両が車線変更を完了する前に自車両が高速で走行して事前に追い抜いた場合に，低速車両の追従とならないのに，採点結果としては採点完了となるのではと思ったためです。

- 関連 issue #34

kasecato, 2023-01-28:
ローカル環境では決して走破できない速度設定でもオンライン環境では採点が完了してしまいます。

kasecato, 2023-01-29:
ローカル環境で走破できない設定でも，オンライン環境で採点完了となれば有効なスコアと判定されますでしょうか？

Reee009876, 2023-01-30:
お世話になっております。

>ローカル環境で走破できない設定でも，オンライン環境で採点完了となれば有効なスコアと判定されますでしょうか？

採点完了と表示されれば、有効なスコアとして判断されます。

※該当箇所がオンライン環境ではどのようになっているかは、ルールの性質上ご案内できません。ご了承ください。

kasecato, 2023-01-30:
承知しました。ご確認ありがとうございました。

closed 2023-01-30 by kasecato