AutomotiveAIChallenge/aichallenge2022-sim #38: オンライン評価でのステータス【結果なし】からの【採点完了】変化について

hidechun, 2023-01-29:
事務局の皆様
お世話になっております。

当方の最新環境をオンライン評価に投入したところ、
投入後１．５時間経過後に【結果なし】表示になった後、更に数時間（詳細な時間は記憶なし。結果なし状態から次評価を行ったかもしれません）経過したところ、【採点完了】並びに評価用シナリオ結果が表示される状況が発生しております。

因みに、以下の60秒待ち対応を入れたものです。
https://github.com/AutomotiveAIChallenge/aichallenge2022-sim/issues/20#issuecomment-1383092438

以下を確認させてください。
　・採点完了となった環境の採点は有効でしょうか。
　・最終の投稿で【結果なし】となった場合は失格でしょうか？
　⇛ルールを見ると失格と読み取れますが、採点が安定しないので【結果なし】が失格確定であれば、最終投稿環境をどうするか熟慮したく聞いておきたいです。失格・記録なしは悲しすぎますので。
　・スコアは途中経過に最速スコアが含まれたとしても、最終投稿の環境スコアとなりますか？
　⇛最終投稿環境をどうするか検討のために念の為確認させてください。

投稿履歴とコメントを添付致します。
ーーーーーーーーーーー
2023-01-30 06:19 | 結果なし |  -- | -- | --　 　⇛　新たな環境を投稿。ローカル環境では走破しています。
2023-01-30 00:09 | 結果なし |  　⇛　19:20投稿と同一環境を投稿。採点完了に遷移する？
2023-01-29 19:20 | 採点完了 | 106.16秒　⇛【結果なし】から【採点完了】に遷移
2023-01-29 17:50 | 採点完了 | 110.40秒　⇛【結果なし】から【採点完了】に遷移
2023-01-29 16:35 | 採点完了 | 109.60秒　⇛【結果なし】から【採点完了】に遷移
2023-01-28 19:01 | 採点完了 | 121.87秒　⇛【採点完了】
ーーーーーーーーーーー
以上です。

hidechun, 2023-01-30:
21:15に確認しましたが。採点完了になっていました。
採点完了になることを信じ、最終版を提出しようと思います。

2023-01-30 06:19 | 採点完了 | 113.74秒
2023-01-30 00:09 | 採点完了 | 109.83秒

Reee009876, 2023-01-30:
お世話になっております。

以下の通り、回答いたします。
・採点完了となった環境の採点は有効でしょうか。
⇒　有効となります、

・最終の投稿で【結果なし】となった場合は失格でしょうか？
⇒　一度、採点完了となった場合、その後の採点で結果なしとなったとしても、採点完了時のスコアが維持されます。
例：一回目：110s、二回目：結果なし　⇒　110sが維持

・順位決定スコアは途中経過に最速スコアが含まれたとしても、最終投稿の環境スコアとなりますか？
⇛　こちら只今確認中です。分かり次第、共有いたします。

Reee009876, 2023-01-30:
申し訳ございません
直前の投稿の確認が漏れていました。

結果なしが採点完了になる事象に関しても
早急に確認いたします。

Kenzo-Fujisaki, 2023-01-31:
@Reee009876 さま
こんにちは。
確認の方はいかがでしょうか？

私達のチームは現在
現在のランキングに掲載されているタイムより
大幅に更新が可能なデータをアップロードする予定であります。

しかし、昨日の「結果なし」がサーバの不具合であることを懸念しているため
アップロードを躊躇しています。

kasecato, 2023-01-31:
こちらもアップロードしたファイルが採点結果なしと表示された為，車両を遅く調整したものをアップロードして本日の 3 回リミットになりました。しかし午前 1 時に投稿した 1 回目の結果が午前 11 時ごろに採点完了となっており，このままでは最後に投稿したものが最終スコアになってしまいそうで，少し後悔しています。
[image]

Reee009876, 2023-01-31:
お世話になっております．

スコアの算出の際に一度の施行ではなく複数回施行しているのですが，
その途中経過が表示されてしまっている状況です．

例:
下記のようになっていた場合，2回目施行の段階でご確認いただくと結果なしと表示され，すべての施行が完了したあとにご確認いただくとスコアありになります．
1回目 結果なし 
2回目 結果なし
x回目 300s 
n回目 200s 

現在，全てが完了次第，スコアが表示されるように修正を加えるようサーバ班に指示を出しております．

ご迷惑をおかけして申し訳ありません．

Kenzo-Fujisaki, 2023-01-31:
@Reee009876 さま
返信ありがとうございます。

途中経過が表示されているということですが
そうなると、結果なしの場合は、いずれにせよ結果無しだと思います。

しかし、今回は結果なしからスコア表示へと遷移されているため。
途中結果が表示されているとは思えないです。

今の所、スコア表示の修正の終了目処はどのくらいでしょうか？

dazzhe, 2023-01-31:
@kasecato 
> こちらもアドバンスト コースに午前1時にアップロードしたファイルが採点結果なしと表示された為，車両を遅く調整したものを午前 8 時にアップロードして本日1月31日の 3 回リミットになりました。しかし午前 1 時に投稿した 1 回目の結果が午前 11 時ごろに採点完了となっており，このままでは最後に投稿したものが最終スコアになってしまいそうで，少し後悔しています。

大変申し訳ございません。サーバー側の不具合による問題ですので、もしご希望されるのでしたら最新の採点結果については削除させていただくことも可能ですが、いかがでしょうか。

kasecato, 2023-01-31:
@dazzhe san
大変お手数をおかけしますが，最後に投稿して 15:52 時点でまだ結果が表示されていない以下のファイルを削除頂ければと思います 🙇‍♂️ アカウントは YutakaJCT です。メールは kasecato で始まります。
- 2023-01-31 02:55 結果なし
- 2023-01-31 08:34 結果なし

[image]

dazzhe, 2023-01-31:
@kasecato さま
承知いたしました。削除後またご連絡させていただきます。

dazzhe, 2023-01-31:
@kasecato さま
- 2023-01-31 02:55 結果なし
- 2023-01-31 08:34 結果なし
の結果について削除いたしました。ご確認ください。

kasecato, 2023-01-31:
@dazzhe san
結果なしとなった投稿が以下のとおり削除されていることを確認しました。ありがとうございました🙇‍♂️

[image]

uhobeike, 2023-01-31:
@Reee009876 @dazzhe 

こんにちは
@Kenzo-Fujisaki と同じチームの一員である池邉龍宏です。
昨日の結果なしは、私達のチームにおいても同様にサーバーのエラーによるもの
なのでしょうか？
[image]

サーバーの状態がわからないため。
今日の残りの時間のうちにアップロードの3回分を消費したいと考えているのですが
サーバーエラーによる「結果なし」
という昨日と同じような結果が返ってくる可能性を懸念しています。

本日の提出は控えるべきなのでしょうか？
それとも提出を行っても問題なく結果が返ってくるのでしょうか？

dazzhe, 2023-01-31:
@uhobeike さま
サーバー確認いたしました。
結果なしと表示されているのはサーバーエラーが原因ではなく、制限速度を超えているため失格扱いになっているようです。

uhobeike, 2023-01-31:
@dazzhe 確認ありがとうございます。
わかりました。

hidechun, 2023-01-31:
投稿期間が過ぎたため、本トピックはCloseします。
最終順位が楽しみですね。

closed 2023-01-31 by hidechun